natanaeljr/esp32-MPU-driver #5: mpu_real example not working with MPU9250

sukeshak, 2018-06-02:
First of all, thank you for the code example... I am trying to get mpu_real working with M5Stack board.
Here is the output which shows 0 for all the roll,pitch & yaw.

$ MPU Driver Example: Advanced
E (2) example: **Through I2C**
I (177) MPU9250: **whoAmI [0x71]:0X71**
I (181) example: MPU connection successful!
I (287) MPU9250: Reset!
I (303) MPU9250: Sample rate set to 100 Hz
I (306) MPU9250: Initialization complete
I (309) MPU9250: Sample rate set to 1000 Hz
W (311) MPU9250: Compass sample rate constrained to 100, magnetometer's maximum is 100 Hz
I (664) MPU9250: Sample rate set to 100 Hz
I (672) MPU9250: Sample rate set to 1000 Hz
W (674) MPU9250: Compass sample rate constrained to 100, magnetometer's maximum is 100 Hz
I (1024) MPU9250: Sample rate set to 100 Hz
I (1030) example: MPU Self-Test result: **Gyro=OK Accel=OK**
I (1032) MPU9250: Sample rate set to 1000 Hz
W (1034) MPU9250: Compass sample rate constrained to 100, magnetometer's maximum is 100 Hz
I (1388) MPU9250: Sample rate set to 100 Hz
I (1396) MPU9250: Sample rate set to 250 Hz
W (1398) MPU9250: Compass sample rate constrained to 83, magnetometer's maximum is 100 Hz
I (1401) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:1
Pitch:   +0.0    Roll:   +0.0    Yaw:   +0.0
Pitch:   +0.0    Roll:   +0.0    Yaw:   +0.0
Pitch:   +0.0    Roll:   +0.0    Yaw:   +0.0
Pitch:   +0.0    Roll:   +0.0    Yaw:   +0.0

Do let me know if you have any suggestions to get this working...
Also does your library support DMP?

natanaeljr, 2018-06-03:
Hey, thanks for you feedback. My guess is that it is somenthing with interrupt pin.

The mpuTask runs only when a interrupt signal occurs. Since you are always getting 0 in all axis, it means mpuTask has not received any interrupt signal notification.

Check if you have the MPU interrupt pin connected to a esp32 GPIO, it should match the GPIO number set in this [line](https://github.com/natanaeljr/esp32-MPU-driver/blob/master/examples/mpu_real/main/mpu_real.cpp#L65).

natanaeljr, 2018-06-03:
DMP support will be added to the library. I am writing the code to enable the various DMP features.

Currently the work is paused, but should I should continue it soon.

sukeshak, 2018-06-03:
Thank you for the response. Will check with them and verify the correct pin. 
The schematics don't have the details.

Good to know DMP is in the to-do list :)

Btw, is it possible to keep the device in any (stable) angle and do calibration, so that all the data (orientation like roll, pitch & yaw) is relative to the initial position?

natanaeljr, 2018-06-03:
The library does not do this sort of calibration on its own.
But it is easy to do it in the application code.
You can simply save the initial position angle and then offset it on next readings, then just do a simple check of the angle range like it's done with [yaw](https://github.com/natanaeljr/esp32-MPU-driver/blob/82a609236b7773ceaafc11f817ca117371314ce2/examples/mpu_real/main/mpu_real.cpp#L228-L232) in mpu_real example.

sukeshak, 2018-06-04:
Thank you for the response. Will try that out.

closed 2018-06-04 by natanaeljr