natanaeljr/esp32-MPU-driver #6: Example mpu_real interrupt isn't working

TurkeyLegsWilliam, 2018-06-15:
I ran your example mpu_real, but it doesn't seem to be working;
particularly, it's getting stuck at the line
`uint32_t notificationValue = ulTaskNotifyTake(pdTRUE, portMAX_DELAY);`
I put some logs in and it looks like mpuISR isn't running.

I was wondering if there was a bug in the code or is it just something wrong on my end? I set GPIO 17 as the interrupt pin, if that helps and the interrupt pin on the MPU seems to output a square wave on an oscilloscope.

natanaeljr, 2018-06-16:
I checked and tested the example code again, it still works. So I guess there might be something wrong with your GPIO 17 in ESP32. Maybe you could try another GPIO, just change the number [here](https://github.com/natanaeljr/esp32-MPU-driver/blob/master/examples/mpu_real/main/mpu_real.cpp#L65).
If that still does not work and you have changed something in the code, please post the changes here.

TurkeyLegsWilliam, 2018-06-18:
Changing the pins does not seem to work. Perhaps it's a configuration error.

sunhaoqin, 2019-01-24:
The frequency is 250hz,i think its too high.because it will occur a interrupt every 4ms.but from line211 to line236 ,it takes about 10ms.this will cause notificationValue>1.maybe you can change the frequency lower than 20hz.it works!

danylook, 2019-06-06:
i had the same problem. i change the mpu and works like a charm