natanaeljr/esp32-MPU-driver #10: Tilt Angles are updated even no rototion or movement of gyroscope(MPU6050)

dhanasekaran26, 2018-11-30:
Hi!

I build and flash the mpu_real it outputs looks below 
Pitch:   -5.5 	 Roll:  +12.0 	 Yaw: -163.8 
Pitch:  -13.6 	 Roll:  +50.1 	 Yaw: -152.3 
Pitch:   -3.2 	 Roll:   +9.2 	 Yaw: -149.0 
Pitch:  -14.6 	 Roll:  +51.0 	 Yaw: -144.7 
Pitch:   -6.2 	 Roll:  +29.5 	 Yaw: -143.7 
Pitch:   -7.8 	 Roll:  +38.5 	 Yaw: -154.8 
Pitch:  -16.9 	 Roll:  +68.5 	 Yaw: -157.9 
Pitch:   -7.7 	 Roll:  +35.0 	 Yaw: -156.6 
Pitch:   -8.8 	 Roll:  +30.8 	 Yaw: -173.8

in this case I am not use ulTaskNotifyTake();

I don't know why it happen's please give your suggestion.

Regards,
K dhanasekaran

natanaeljr, 2018-11-30:
Please, post the modifications you have done to the example code

dhanasekaran26, 2018-12-03:
Hi!
sorry for the delay and I did not get any notification from MPUISR() so just skip it as blew, 

   // Wait for notification from mpuISR
      //  uint32_t notificationValue = ulTaskNotifyTake(pdTRUE, portMAX_DELAY);

        //uint32_t notificationValue = ulTaskNotifyTake(pdTRUE, 100);

        uint32_t notificationValue = 0; 
        if (notificationValue > 1) {
            ESP_LOGW(TAG, "Task Notification higher than 1, value: %d", notificationValue);
            MPU.resetFIFO();
            continue;
        }

Regards,
K dhanasekaran

dhanasekaran26, 2018-12-04:
Ok, now I am using MPU_I2C to read raw data manually and then use your calculation to find pitch ,roll and yaw. Can you tell me how to change the range of pitch to 0 - 360 , and also others range  and where can I find explanation for those calculation.

Thanks and Regards,
Dhanasekaran

closed 2019-01-02 by natanaeljr