natanaeljr/esp32-MPU-driver #26: How to use this driver in c project? e.g. an ESP-IDF hello_world example project written in C language

sonictl, 2022-02-09:
## How to use this driver in C project? , e.g. an ESP-IDF hello_world example project written in C language

Hello, I tried to import the i2c and MPU repository as component under my 'hello_world' example project given by ESP-IDF.
if I modify the main.c into main.cpp by modifying some lines in `main.c` file and `CMakeList.txt`, the components works.

However, My project with lvgl lib needs to be developed by C language. When I directly using this driver for my MPU6050 on I2C, the error occurs when compiling.

**How to use the cpp-written components in C project?**

below is the file structure of my project:

```
% tree -L 3
.
├── CMakeLists.txt
├── Makefile
├── README.md
├── components
│   ├── I2Cbus
│   │   ├── CMakeLists.txt
│   │   ├── Kconfig
│   │   ├── Kconfig.projbuild
│   │   ├── LICENSE
│   │   ├── Makefile.projbuild
│   │   ├── README.md
│   │   ├── component.mk
│   │   ├── examples
│   │   ├── include
│   │   └── src
│   └── MPUdriver
│       ├── CMakeLists.txt
│       ├── Kconfig
│       ├── Kconfig.projbuild
│       ├── LICENSE
│       ├── Makefile.projbuild
│       ├── README.md
│       ├── component.mk
│       ├── docs
│       ├── examples
│       ├── include
│       ├── src
│       ├── test
│       └── tools
├── example_test.py
├── main
│   ├── CMakeLists.txt
│   ├── component.mk
│   ├── hello_world_main.c.bak
│   ├── main.c
│   └── main.cpp.ok.bak
├── sdkconfig
└── sdkconfig.old
```

sonictl, 2022-02-09:
The error info given by compiler:

```
In file included from ../components/MPUdriver/include/MPU.hpp:39,
                 from ../main/main.c:26:
../components/I2Cbus/include/I2Cbus.hpp:36:1: error: unknown type name 'namespace'
 namespace i2cbus {
 ^~~~~~~~~
../components/I2Cbus/include/I2Cbus.hpp:36:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 namespace i2cbus {
                  ^
../components/I2Cbus/include/I2Cbus.hpp:43:1: error: unknown type name 'using'; did you mean 'ulong'?
 using I2C_t = i2cbus::I2C;
 ^~~~~
 ulong
../components/I2Cbus/include/I2Cbus.hpp:43:15: error: 'i2cbus' undeclared here (not in a function)
 using I2C_t = i2cbus::I2C;
               ^~~~~~
../components/I2Cbus/include/I2Cbus.hpp:43:21: error: expected ',' or ';' before ':' token
 using I2C_t = i2cbus::I2C;
                     ^
../components/I2Cbus/include/I2Cbus.hpp:46:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'i2c0'
 extern I2C_t i2c0;        /*!< port: I2C_NUM_0 */
              ^~~~
../components/I2Cbus/include/I2Cbus.hpp:47:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'i2c1'
 extern I2C_t i2c1;        /*!< port: I2C_NUM_1 */
              ^~~~
../components/I2Cbus/include/I2Cbus.hpp:51:1: error: unknown type name 'namespace'
 namespace i2cbus {
 ^~~~~~~~~
../components/I2Cbus/include/I2Cbus.hpp:51:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 namespace i2cbus {
                  ^
In file included from ../components/MPUdriver/include/MPU.hpp:39,
                 from ../main/main.c:26:
../components/I2Cbus/include/I2Cbus.hpp:159:1: error: unknown type name 'constexpr'; did you mean 'const'?
 constexpr I2C_t& getI2C(i2c_port_t port) {
 ^~~~~~~~~
 const
../components/I2Cbus/include/I2Cbus.hpp:159:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before '&' token
 constexpr I2C_t& getI2C(i2c_port_t port) {
                ^
In file included from ../components/MPUdriver/include/mpu/types.hpp:16,
                 from ../components/MPUdriver/include/MPU.hpp:52,
                 from ../main/main.c:26:
../components/MPUdriver/include/mpu/registers.hpp:19:1: error: unknown type name 'namespace'
 namespace mpud
 ^~~~~~~~~
../components/MPUdriver/include/mpu/registers.hpp:20:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from ../components/MPUdriver/include/MPU.hpp:52,
                 from ../main/main.c:26:
../components/MPUdriver/include/mpu/types.hpp:20:1: error: unknown type name 'namespace'
 namespace mpud
 ^~~~~~~~~
../components/MPUdriver/include/mpu/types.hpp:21:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from ../main/main.c:26:
../components/MPUdriver/include/MPU.hpp:55:1: error: unknown type name 'namespace'
 namespace mpud
 ^~~~~~~~~
../components/MPUdriver/include/MPU.hpp:56:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
../components/MPUdriver/include/MPU.hpp:61:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 typedef mpud::MPU MPU_t;
             ^
../components/MPUdriver/include/MPU.hpp:63:1: error: unknown type name 'namespace'
 namespace mpud
 ^~~~~~~~~
../components/MPUdriver/include/MPU.hpp:64:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
../components/MPUdriver/include/MPU.hpp:263:1: error: unknown type name 'namespace'
 namespace mpud
 ^~~~~~~~~
../components/MPUdriver/include/MPU.hpp:264:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from ../main/main.c:27:
../components/MPUdriver/include/mpu/math.hpp:21:1: error: unknown type name 'namespace'
 namespace mpud
 ^~~~~~~~~
../components/MPUdriver/include/mpu/math.hpp:22:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
../main/main.c: In function 'app_main':
../main/main.c:39:12: error: 'myI2C' undeclared (first use in this function)
     I2C_t& myI2C = i2c0;  // i2c0 and i2c1 are the default objects
            ^~~~~
../main/main.c:39:12: note: each undeclared identifier is reported only once for each function it appears in
../main/main.c:39:20: error: 'i2c0' undeclared (first use in this function)
     I2C_t& myI2C = i2c0;  // i2c0 and i2c1 are the default objects
                    ^~~~

```

Ishan-Sharma-17, 2025-02-14:
Did you ever get it to work?